Seeed-Studio/Grove_Temper_Humidity #2: TH02 does not work on Intel Edison

gustavotemple, 2017-09-26:
Hello,

@jeferrb and me, we are trying to use the TH02 on Intel Edison, but in very cold enviroments this code below does not work after few minutes:

```
#include <Wire.h>
#include <TH02_dev.h>
#include "rgb_lcd.h"

rgb_lcd lcd;

int counter = 0;

void setup() {
	lcd.begin(16, 2);
}

void loop() {
	float temperature = TH02.ReadTemperature();
	lcd.clear();
	lcd.print("loop() [");
	lcd.print(++counter);
	lcd.print("]");
	lcd.setCursor(0, 1);
	lcd.print(" - Temp: ");
	lcd.print(temperature);
	lcd.print(" C");
}
```

There is sometime related with this issue below?
https://github.com/intel-iot-devkit/upm/issues/351

KillingJacky, 2017-10-16:
Hi, sorry for the late reply, do you still have this issue now? If so, what kind of "not work" do you mean? program get stuck? temperature not accurate?

gustavotemple, 2017-10-16:
Hello @KillingJacky,
Yes, the program get stuck

KillingJacky, 2017-10-17:
OK, program getting stuck means the I2C reading fails all the time, and the library falls into a while true loop. Typically this may happen when the sensor is placed to work under critical environment. From your description, `very cold enviroments (< 24°C) `, 24°C can't be said very cold, can I suppose it to be -24°C? If so, from the specification of this sensor, it only works under:
```
Humidity: 0% - 80% RH 
Temperature: 0 ~ 70 °C
```
Please refer to http://wiki.seeed.cc/Grove-TemptureAndHumidity_Sensor-High-Accuracy_AndMini-v1.0/

gustavotemple, 2017-10-30:
Hello @KillingJacky,

Sorry, I fixed the description, I live in Brazil, so +22°C is very cold for us.

This was the answer from the Intel Corporation:

https://communities.intel.com/message/500271

closed 2021-09-18 by Pillar1989