Seeed-Studio/Grove_Temper_Humidity #8: 'TH02' was not declared in this scope

little-dave95, 2022-01-06:
Hi,

I have an issue with this Sketch:

Arduino: 1.8.19 (Mac OS X), Board: "Arduino Uno WiFi Rev2, ATMEGA328"

'TH02' was not declared in this scope....

I'm quite new to coding so I don't know what else you need....

here is the full sketch...

#include <TH02_dev.h>    
#include "Arduino.h"
#include "Wire.h"    
void setup()
{
Serial.begin(9600); // start serial for output
Serial.println("****TH02_dev demo by seeed studio****\n");
/* Power up,delay 150ms,until voltage is stable */
delay(150);
/* Reset HP20x_dev */
TH02.begin();
delay(100);
/* Determine TH02_dev is available or not */
Serial.println("TH02_dev is available.\n");
}
void loop() {
        
float temper = TH02.ReadTemperature();
Serial.println("Temperature: ");
Serial.print(temper);
Serial.println("C\r\n");
float humidity = TH02.ReadHumidity();
Serial.println("Humidity: ");
Serial.print(humidity);
Serial.println("%\r\n");
delay(1000);
 
}

imamnurby, 2022-02-17:
I think you can define add  `TH02_dev TH02;` before void setup()

lakshanthad, 2022-02-20:
> I think you can add `TH02_dev TH02;` before void setup()
> 
> *edit due to grammar

Did this solution work for you @little-dave95? I am going to close this issue if that is okay.

Lesords, 2024-10-16:
Hello,

I'm going to close this issue, feel free to re-open it if you have any other questions.

closed 2024-10-16 by Lesords