Seeed-Studio/Grove_Temper_Humidity #9: Support for Raspberry Pi?

phiresky, 2022-01-29:
Is it possible to use this module with Raspberry Pi Base Hats?

Lesords, 2024-10-16:
Hello,

I'm very sorry to have kept you waiting so long.

Do you still have this problem now?

phiresky, 2024-10-16:
Hey! I haven't used this module for a while now, so not directly. I still think it's an interesting questions, though I guess it applies to the whole ecosystem and not just this one sensor.

Lesords, 2024-10-16:
I don't have this hardware device at the moment, I will verify it once I have it and let you know the result

Lesords, 2025-01-06:
Hello，

Sorry to keep you waiting again, I've been a bit busy lately.

I just checked, the current repository does not support use on Raspberry Pi.

But you can use grove sensors on Raspberry Pi via grove.py.

Read the documentation and repositories below.

Wiki：https://wiki.seeedstudio.com/Grove_Base_Hat_for_Raspberry_Pi/
Repo：https://github.com/Seeed-Studio/grove.py

closed 2025-01-06 by Lesords