Seeed-Studio/Grove_Temper_Humidity #14: Error on compilation on intel genuino 101

maisotro, 2025-03-27:
When a try to compile the example (TH02_dev demo )on a intel genuino 101 board  

```cpp
/home/celito/Arduino/libraries/Grove_Temper_Humidity_TH02/THSensor_base.cpp: In member function ‘uint16_t TempHumi::IIC_ReadData2byte(uint8_t, uint8_t)’:
/home/celito/Arduino/libraries/Grove_Temper_Humidity_TH02/THSensor_base.cpp:60:32: error: variable-sized object ‘tmpArray’ may not be initialized
     uint16_t tmpArray[num] = {0};
                                ^

exit status 1

Compilation error: exit status 1
```

Lesords, 2025-03-28:
Hello, 

Thank you for raising this question. 

At present, it seems that this repository has not been adapted for the Genuino 101, and there will indeed be this compilation issue (in addition to undefined problems). 

You can avoid the compilation issue by changing the line `uint16_t tmpArray[num] = {0};` in the file `/home/celito/Arduino/libraries/Grove_Temper_Humidity_TH02/THSensor_base.cpp` to `uint16_t tmpArray[num];`.

maisotro, 2025-03-31:
With the correction that you provide, i was able to run on the genuino

closed 2025-04-25 by github-actions[bot]